sychaichangkun/ROS-Academy-for-Beginners #45: 与orb2slam的对接

YuYuCong, 2018-04-26:
在运行launch是找不到RGBD节点
[image]

确认已经添加了pkg的路径，以下是~/.bashrc的部分内容
[image]

YuYuCong, 2018-04-26:
QUESTION

sychaichangkun, 2018-04-27:
确定编译过ORB_SLAM2的ROS版吗？
可以看看ORB_SLAM2/Examples/ROS/ORB_SLAM2里有没有RGBD，Mono, Stereo这些二进制文件
如果有的话可能是.bashrc里的ROS_PACKAGE_PATH环境变量没有加上

YuYuCong, 2018-04-30:
检查了ORB_SLAM2/Examples/ROS/ORB_SLAM2里面确实没有RGBD，Mono, Stereo这些二进制文件。
然后我记起来./build_ros.sh编译的时候有报错信息：
请问以下报错您是否已有解决方案？
[image]

sychaichangkun, 2018-05-02:
应该是没有链接boost，看看这个
[https://stackoverflow.com/questions/25864797/error-adding-symbols-dso-missing-from-command-line](url)

YuYuCong, 2018-05-05:
非常感谢！问题已经得到了解决！
#问题描述：
编译./build_ros.sh时报错：
/usr/lib/x86_64-linux-gnu/libboost_system.so: error adding symbols: DSO missing from command line
#解决方案：
在ORB_SLAM2/Examples/ROS/ORB_SLAM2/Cmakelist.txt 文件后面加入两句：
target_link_libraries(Stereo boost_system boost_filesystem)  
target_link_libraries(RGBD boost_system boost_filesystem)  
保存后再次编译即可。

closed 2018-05-05 by YuYuCong