sychaichangkun/ROS-Academy-for-Beginners #50: Invoking "make cmake_check_build_system" failed

daniao2017, 2018-07-25:
用那个推荐的编译器写基础教程时出现的问题！！！！！！！！！！
就是那啥catkin_ws过不了
CMake Error at ROS-Academy-for-Beginners-master/topic_demo/CMakeLists.txt:154 (add_executable):
  add_executable cannot create target "talker" because another target with
  the same name already exists.  The existing target is an executable created
  in source directory "/home/daniao/catkin_ws/src/beginner_tutorials".  See
  documentation for policy CMP0002 for more details.

CMake Error at ROS-Academy-for-Beginners-master/topic_demo/CMakeLists.txt:157 (target_link_libraries):
  Attempt to add link library "/opt/ros/kinetic/lib/libroscpp.so" to target
  "talker" which is not built in this directory.

CMake Error at ROS-Academy-for-Beginners-master/topic_demo/CMakeLists.txt:157 (target_link_libraries):
  Attempt to add link library
  "/usr/lib/x86_64-linux-gnu/libboost_filesystem.so" to target "talker" which
  is not built in this directory.

CMake Error at ROS-Academy-for-Beginners-master/topic_demo/CMakeLists.txt:157 (target_link_libraries):
  Attempt to add link library "/usr/lib/x86_64-linux-gnu/libboost_signals.so"
  to target "talker" which is not built in this directory.

CMake Error at ROS-Academy-for-Beginners-master/topic_demo/CMakeLists.txt:157 (target_link_libraries):
  Attempt to add link library "/opt/ros/kinetic/lib/librosconsole.so" to
  target "talker" which is not built in this directory.

CMake Error at ROS-Academy-for-Beginners-master/topic_demo/CMakeLists.txt:157 (target_link_libraries):
  Attempt to add link library "/opt/ros/kinetic/lib/librosconsole_log4cxx.so"
  to target "talker" which is not built in this directory.

CMake Error at ROS-Academy-for-Beginners-master/topic_demo/CMakeLists.txt:157 (target_link_libraries):
  Attempt to add link library
  "/opt/ros/kinetic/lib/librosconsole_backend_interface.so" to target
  "talker" which is not built in this directory.

CMake Error at ROS-Academy-for-Beginners-master/topic_demo/CMakeLists.txt:157 (target_link_libraries):
  Attempt to add link library "/usr/lib/x86_64-linux-gnu/liblog4cxx.so" to
  target "talker" which is not built in this directory.

CMake Error at ROS-Academy-for-Beginners-master/topic_demo/CMakeLists.txt:157 (target_link_libraries):
  Attempt to add link library "/usr/lib/x86_64-linux-gnu/libboost_regex.so"
  to target "talker" which is not built in this directory.

CMake Error at ROS-Academy-for-Beginners-master/topic_demo/CMakeLists.txt:157 (target_link_libraries):
  Attempt to add link library "/opt/ros/kinetic/lib/libxmlrpcpp.so" to target
  "talker" which is not built in this directory.

CMake Error at ROS-Academy-for-Beginners-master/topic_demo/CMakeLists.txt:157 (target_link_libraries):
  Attempt to add link library
  "/opt/ros/kinetic/lib/libroscpp_serialization.so" to target "talker" which
  is not built in this directory.

CMake Error at ROS-Academy-for-Beginners-master/topic_demo/CMakeLists.txt:157 (target_link_libraries):
  Attempt to add link library "/opt/ros/kinetic/lib/librostime.so" to target
  "talker" which is not built in this directory.

CMake Error at ROS-Academy-for-Beginners-master/topic_demo/CMakeLists.txt:157 (target_link_libraries):
  Attempt to add link library "/opt/ros/kinetic/lib/libcpp_common.so" to
  target "talker" which is not built in this directory.

CMake Error at ROS-Academy-for-Beginners-master/topic_demo/CMakeLists.txt:157 (target_link_libraries):
  Attempt to add link library "/usr/lib/x86_64-linux-gnu/libboost_system.so"
  to target "talker" which is not built in this directory.

CMake Error at ROS-Academy-for-Beginners-master/topic_demo/CMakeLists.txt:157 (target_link_libraries):
  Attempt to add link library "/usr/lib/x86_64-linux-gnu/libboost_thread.so"
  to target "talker" which is not built in this directory.

CMake Error at ROS-Academy-for-Beginners-master/topic_demo/CMakeLists.txt:157 (target_link_libraries):
  Attempt to add link library "/usr/lib/x86_64-linux-gnu/libboost_chrono.so"
  to target "talker" which is not built in this directory.

CMake Error at ROS-Academy-for-Beginners-master/topic_demo/CMakeLists.txt:157 (target_link_libraries):
  Attempt to add link library
  "/usr/lib/x86_64-linux-gnu/libboost_date_time.so" to target "talker" which
  is not built in this directory.

CMake Error at ROS-Academy-for-Beginners-master/topic_demo/CMakeLists.txt:157 (target_link_libraries):
  Attempt to add link library "/usr/lib/x86_64-linux-gnu/libboost_atomic.so"
  to target "talker" which is not built in this directory.

CMake Error at ROS-Academy-for-Beginners-master/topic_demo/CMakeLists.txt:157 (target_link_libraries):
  Attempt to add link library "/usr/lib/x86_64-linux-gnu/libpthread.so" to
  target "talker" which is not built in this directory.

CMake Error at ROS-Academy-for-Beginners-master/topic_demo/CMakeLists.txt:157 (target_link_libraries):
  Attempt to add link library
  "/usr/lib/x86_64-linux-gnu/libconsole_bridge.so" to target "talker" which
  is not built in this directory.

daniao2017, 2018-07-26:
老哥，那个网站登录不了啊～～～～
要邀请呢～～～～

lazyparser, 2018-07-26:
按照提示给管理员发邮件就行。

closed 2019-04-25 by sychaichangkun