sychaichangkun/ROS-Academy-for-Beginners #54: melodic cmake 出现问题

1miaoting1, 2018-10-10:
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "hector_mapping"
  with any of the following names:

    hector_mappingConfig.cmake
    hector_mapping-config.cmake

  Add the installation prefix of "hector_mapping" to CMAKE_PREFIX_PATH or set
  "hector_mapping_DIR" to a directory containing one of the above files.  If
  "hector_mapping" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  ROS-Academy-for-Beginners/slam_sim_demo/CMakeLists.txt:3 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/mt/tutorial_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/mt/tutorial_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

1miaoting1, 2018-10-10:
装依赖的时候就有问题。花了不少时间解决了。但是cmake又出问题。好难受。。。我是不是应该把melodic换成kinetic?

1miaoting1, 2018-10-10:
说错了。是catkin_make出错。

lazyparser, 2018-10-13:
ROS的版本跟ubuntu版本对应的，请安装对应的版本。

这里是缺少hector的相关算法包。按照ros网站的信息安装即可。

后续还有相关问题可以去q站提问，那里有顾问负责回答。 https://q.droid.ac.cn 从置顶的帖子中发送邮件邀请注册。

closed 2019-06-10 by sychaichangkun