sychaichangkun/ROS-Academy-for-Beginners #62: 无法得到模型

YCH188, 2019-04-11:
环境：ubuntu16.04
ros：kinetic
gazebo：8.6.0
程序都已经编译好了，运行roslaunch robot_sim_demo robot_spawn.launch
无法得到正确的模型，如图

[image]
---------------------------
log如下：
```
... logging to /home/ubuntu/.ros/log/36ae25a6-5c03-11e9-b72e-00117f13914c/roslaunch-ubuntu-OptiPlex-960-24353.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
xacro.py is deprecated; please use xacro instead
started roslaunch server http://ubuntu-OptiPlex-960:35975/

SUMMARY
========

PARAMETERS
 * /cmd_vel_mux/yaml_cfg_file: /home/ubuntu/test...
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_sim_time: True
 * /xbot/joint_state_controller/publish_rate: 50
 * /xbot/joint_state_controller/type: joint_state_contr...
 * /xbot/pitch_platform_position_controller/joint: yaw_to_pitch_plat...
 * /xbot/pitch_platform_position_controller/pid/d: 10.0
 * /xbot/pitch_platform_position_controller/pid/i: 0.01
 * /xbot/pitch_platform_position_controller/pid/p: 100.0
 * /xbot/pitch_platform_position_controller/type: effort_controller...
 * /xbot/robot_state_publisher/publish_frequency: 20.0
 * /xbot/yaw_platform_position_controller/joint: base_to_yaw_platform
 * /xbot/yaw_platform_position_controller/pid/d: 10.0
 * /xbot/yaw_platform_position_controller/pid/i: 0.01
 * /xbot/yaw_platform_position_controller/pid/p: 100.0
 * /xbot/yaw_platform_position_controller/type: effort_controller...

NODES
  /xbot/
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    spawner (controller_manager/spawner)
  /
    cmd_vel_mux (nodelet/nodelet)
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    mobile_base_nodelet_manager (nodelet/nodelet)
    urdf_spawner (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [24366]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 36ae25a6-5c03-11e9-b72e-00117f13914c
process[rosout-1]: started with pid [24379]
started core service [/rosout]
process[gazebo-2]: started with pid [24399]
process[gazebo_gui-3]: started with pid [24408]
process[urdf_spawner-4]: started with pid [24413]
process[xbot/spawner-5]: started with pid [24414]
process[xbot/robot_state_publisher-6]: started with pid [24415]
process[mobile_base_nodelet_manager-7]: started with pid [24416]
process[cmd_vel_mux-8]: started with pid [24426]
[INFO] [1554950450.420971, 0.000000]: Controller Spawner: Waiting for service controller_manager/load_controller
[ INFO] [1554950450.494971468]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1554950450.496213075]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1554950450.603158936]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1554950450.604651437]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
SpawnModel script started
[INFO] [1554950450.984909, 0.000000]: Loading model XML from ros parameter
[INFO] [1554950450.989013, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[INFO] [1554950451.594650, 0.000000]: Calling service /gazebo/spawn_urdf_model
Warning [parser.cc:838] XML Element[mu], child of element[link] not defined in SDF. Ignoring[mu]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:838] XML Element[mu], child of element[link] not defined in SDF. Ignoring[mu]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:838] XML Element[mu], child of element[link] not defined in SDF. Ignoring[mu]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:838] XML Element[mu], child of element[link] not defined in SDF. Ignoring[mu]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
Warning [parser.cc:838] XML Element[mu], child of element[link] not defined in SDF. Ignoring[mu]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
[INFO] [1554950453.757049, 0.001000]: Spawn status: SpawnModel: Successfully spawned entity
[ INFO] [1554950453.907029952, 0.001000000]: Loading gazebo_ros_control plugin
[ INFO] [1554950453.907467380, 0.001000000]: Starting gazebo_ros_control plugin in namespace: /xbot
[ INFO] [1554950453.909376902, 0.001000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [/robot_description] on the ROS param server.
[urdf_spawner-4] process has finished cleanly
log file: /home/ubuntu/.ros/log/36ae25a6-5c03-11e9-b72e-00117f13914c/urdf_spawner-4*.log
[ WARN] [1554950454.031958429, 0.001000000]: Deprecated syntax, please prepend 'hardware_interface/' to 'EffortJointInterface' within the <hardwareInterface> tag in joint 'base_to_yaw_platform'.
[ WARN] [1554950454.033772617, 0.001000000]: Deprecated syntax, please prepend 'hardware_interface/' to 'EffortJointInterface' within the <hardwareInterface> tag in joint 'yaw_to_pitch_platform'.
[INFO] [1554950454.046065, 0.001000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[ INFO] [1554950454.048094734, 0.001000000]: Loaded gazebo_ros_control.
[INFO] [1554950454.049199, 0.001000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1554950454.051939, 0.002000]: Loading controller: joint_state_controller
[INFO] [1554950454.069773, 0.012000]: Loading controller: yaw_platform_position_controller
[ INFO] [1554950454.084982113, 0.022000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1554950454.086037083, 0.022000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1554950454.273298212, 0.161000000]: Physics dynamic reconfigure ready.
[ INFO] [1554950454.322919869, 0.203000000]: Physics dynamic reconfigure ready.
[INFO] [1554950454.355930, 0.230000]: Loading controller: pitch_platform_position_controller
[INFO] [1554950454.455358, 0.315000]: Controller Spawner: Loaded controllers: joint_state_controller, yaw_platform_position_controller, pitch_platform_position_controller
[INFO] [1554950454.459460, 0.317000]: Started controllers: joint_state_controller, yaw_platform_position_controller, pitch_platform_position_controller
^C[cmd_vel_mux-8] killing on exit
[mobile_base_nodelet_manager-7] killing on exit
[xbot/spawner-5] killing on exit
[xbot/robot_state_publisher-6] killing on exit
[gazebo-2] killing on exit
[gazebo_gui-3] killing on exit
[INFO] [1554950673.869067, 216.056000]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1554950673.870748, 216.058000]: Stopping all controllers...
[INFO] [1554950673.880746, 216.063000]: Unloading all loaded controllers...
[INFO] [1554950673.881388, 216.064000]: Trying to unload pitch_platform_position_controller
[INFO] [1554950673.900128, 216.074000]: Succeeded in unloading pitch_platform_position_controller
[INFO] [1554950673.900967, 216.076000]: Trying to unload yaw_platform_position_controller
[INFO] [1554950673.921978, 216.083000]: Succeeded in unloading yaw_platform_position_controller
[INFO] [1554950673.922512, 216.083000]: Trying to unload joint_state_controller
[INFO] [1554950673.928146, 216.087000]: Succeeded in unloading joint_state_controller
```

YCH188, 2019-04-11:
@sychaichangkun 可以帮忙看一下吗？谢谢！

sychaichangkun, 2019-04-25:
怀疑可能是和gazebo版本有关，有些模型没有渲染出来
你用的是gazebo8.x，这个仿真没有在8上测试过
建议用gazebo7来运行

YCH188, 2019-04-25:
@sychaichangkun OK，我之后再测试一下

closed 2019-04-30 by sychaichangkun

YCH188, 2019-05-07:
是gazebo版本的问题，换成7就能显示了，谢谢！

SUTLZY, 2020-10-21:
不改版本，怎么修改呢？