sychaichangkun/ROS-Academy-for-Beginners #66: Can this program run in ROS melodic version?

nTnZone, 2019-07-09:
I use ubuntu18.04 and gazebo 9

lazyparser, 2019-07-09:
perhaps.

Try replace all `kinetic` with `melodic`.

Further errors & problems could be reported here as new issues.

crankler, 2020-02-13:
I have try this package in melodic, but when I use the command: 
`roslaunch robot_sim_demo  robot_spawn.launch`
only ground show up.
Can you tell  me how to fix the world package? Thank you!

mfj710, 2020-03-03:
> I have try this package in melodic, but when I use the command:
> `roslaunch robot_sim_demo robot_spawn.launch`
> only ground show up.
> Can you tell me how to fix the world package? Thank you!

me to,i use ubuntu18.04 and gazebo 9,with hardware nvidia tx2, only ground show up.

anchuanxu, 2020-03-03:
> > I have try this package in melodic, but when I use the command:
> > `roslaunch robot_sim_demo robot_spawn.launch`
> > only ground show up.
> > Can you tell me how to fix the world package? Thank you!
> 
> me to,i use ubuntu18.04 and gazebo 9,with hardware nvidia tx2, only ground show up.

Hi First, you should replace the melodic version of the package with the official link address:
[https://github.com/DroidAITech/ROS-Academy-for-Beginners/tree/melodic](https://github.com/DroidAITech/ROS-Academy-for-Beginners/tree/melodic),This package branch is based on ubuntu18.04 and ROS melodic，Fixed some bugs。If problems persist, please comment below and the package will be maintained continuously.

better-fcl, 2020-04-10:
> I have try this package in melodic, but when I use the command:
> `roslaunch robot_sim_demo robot_spawn.launch`
> only ground show up.
> Can you tell me how to fix the world package? Thank you!

I had the same problem with u. 
You can try the method proposed below by anchuanxu