sychaichangkun/ROS-Academy-for-Beginners #70: 您好，希望能更新imu模型数据

happybored, 2020-03-25:
您好，我们迫切希望能仿真vins-mono框架
但是需要更加详细的imu数据，所需数据如下所示
如果能回复，不胜感激！
#imu parameters       The more accurate parameters you provide, the better performance
acc_n: 0.04          # accelerometer measurement noise standard deviation. #0.2   0.04
gyr_n: 0.004         # gyroscope measurement noise standard deviation.     #0.05  0.004
acc_w: 0.0004         # accelerometer bias random work noise standard deviation.  #0.02
gyr_w: 2.0e-5       # gyroscope bias random work noise standard deviation.     #4.0e-5
g_norm: 9.80766     # gravity magnitude

lazyparser, 2020-03-26:
收到，我们来看下。

anchuanxu, 2020-03-26:
你好 目前仿真xbot上搭载的imu暂时不能仿真vins，如果想仿真vins-mono框架，可以采用自己调整xbot-u.gazebo文件中camera的参数，将其修改为类似d415或者小觅相机参数即可，或者单纯修改现在xbot-u仿真搭载的imu参数。