sychaichangkun/ROS-Academy-for-Beginners #71: confusing code: q.setRPY(yaw,pitch,roll);

fridayfang, 2020-06-14:
I found confusing code in  `ROS-Academy-for-Beginners/tf_demo/src/coordinate_transformation.cpp`:

`q.setRPY(yaw,pitch,roll);`

But it should be set by sequence `roll,yaw,pitch`

anchuanxu, 2020-06-15:
Hi, welcome to ask your question, this question is because in our code we use the Frame of OPENGL, in which the order is yaw,pitch,roll.

sun0912704, 2020-06-15:
thank you. l will learn it three weeks later.

anchuanxu, 2020-06-15:
> thank you. l will learn it three weeks later. 
> […](#)

you are welcome! Welcome to exchange more ros questions

anchuanxu, 2020-06-15:
> > thank you. l will learn it three weeks later. 
> > […](#)
> 
> you are welcome! Welcome to exchange more ros questions

Sorry, I checked the documentation and found that the correct order of this function call is indeed roll pitch yaw, which caused us to code irregularities. Thank you for pointing out that we will correct it in subsequent versions.   
http://docs.ros.org/jade/api/tf/html/c++/classtf_1_1Quaternion.html#ab34dfef6a01c1ece07bea92d485b0576

fridayfang, 2020-06-16:
thanks for your response!!

closed 2020-06-16 by fridayfang